MarkusLange/Teensy_3.x_4.x_and_LC_LIN_Master #1: Compiling Error on Teensy 3.6

jasc82, 2020-02-04:
Hello Markus,

I've tried to use your library on a Teensy 3.6. Unfortunately I'm getting an error when I try to compile it.
Even with the  examples that come with your library. I've attached an txt file with the error message.

It seems that there is a problem in declaration of the serial interfaces. But i'm not experienced enough for further analysis 
Is it possilbe that you take look into it?

Thanks in advance :-)

[Compiling_error_on_Teensy36.txt](https://github.com/MarkusLange/Teensy_3.x_LIN_Master/files/4153716/Compiling_error_on_Teensy36.txt)

MarkusLange, 2020-06-09:
Sorry for my delayed answer did you have some more information for me?
Arduino and Teensyduino version?

I have no compiling error with the Teensy 3.6

jasc82, 2020-06-17:
Hello Markus,

I've updated to Ardunio 1.8.12 and Teensyduino 1.52 and tried it again with the example "NCV7430_Get_FUll-Status".
But I'm still getting the same error:
"C:\Users\schneija\sketchbook\libraries\Teensy_3.x_LIN_Master-master\src\lin_bus.cpp:300:7: error: 'stream' was not declared in this scope"

To proceed in my project I'v used a little workaround by commenting out lines 306 to 309 and line 312 in "lin_bus.cpp". This of course deactivates Serial6 for LIN, but I did not use it anyways.

But I've discovered another problem with your library.
If the slave is not responding it will cause an infite loop and the whole programm will get stuck.

I've implemented a time-out counter to fix this issue.

I've attached my fixes/changes in following TXT file as I'm not so familiar with github.
There are some german comments in the first lines from my side, but guessing from your name this should not be an issue :-)

[lin_bus.cpp.fix_V3.txt](https://github.com/MarkusLange/Teensy_3.x_LIN_Master/files/4791282/lin_bus.cpp.fix_V3.txt)

MarkusLange, 2020-06-26:
On the first place it isn't the actual lib in this github, my fault, the latest on my desk hasn't the Uart behavior on Teensy 3.6 with Uart 6, secondly, I was working on an Timeout fix, your counter has the problem, it depends on how fast your Teensy runs, how log it takes to get an timeout, atm I try different version of an Timeout checks. I have to rewrite some parts to get the Teensy 4.x integrate in it. I hope I will get this soon ready

MarkusLange, 2020-07-02:
New Version is Online if you want give it a try

closed 2020-08-09 by MarkusLange