MarkusLange/Teensy_3.x_4.x_and_LC_LIN_Master #2: No Break Field in Teensy 4

GitSchorschGit, 2021-01-12:
I tested the LIN-Bus on Teensy 4 and I got no Break Field.

BTW, a very good job! 

Best regards

JamesWGT3, 2021-11-04:
Hi, I too have this same issue on a Teensy 4.0. The rest of the message is bang on, but nothing recognises it because the break field is missing.

Please find attached a Pico Scope UART conversion of the Teensy tx pin going to the LIN transceiver of your example file 'NCV7430_Get_Full_Status.ino' to illustrate the issue.

Any help with this would be much appreciated!

[image]

MarkusLange, 2021-11-13:
Very nice Chart where did you get it from. Problem solved, Hardware-UARTS had to shift around. Please test again

closed 2021-11-13 by MarkusLange

JamesWGT3, 2021-11-13:
I made it from the data being transmitted from your library whilst decoding the message in picoscope. It’s a screen grab with screenpresso which then has a nice edit and annotate feature - I’d really recommend the software, I use it on all my pc’s daily.

Thank you for looking into this for me - I’ll try to get the code tested later today and get back to you!

Kind regards,

James