MarkusLange/Teensy_3.x_4.x_and_LC_LIN_Master #6: Wrong enhanced checksum

SteB8Pro, 2022-08-17:
Hello, I think I found a mistake. When using enhanced checksum, the checksum must be calculated using the protected id and not the raw id. I fixed this simply replacing PID with send_pid in line 424.
Sorry but I am completely new to GitHub so I do not know if and how to make the correction on the repository. On my PC I did the modification I described above and the problem was solved.

MarkusLange, 2023-02-15:
Oh yes you are right
[Link](https://microchipdeveloper.com/lin:protocol-dll-lin-message-frame)
[image]
[image]
It says the PID not the FID, correct it thank you

MarkusLange, 2023-02-15:
Closed thanks

closed 2023-02-15 by MarkusLange